AFM-SPM/TopoStats #958: Ability to set image_set for filters, grains and tracing separately to reduce output

llwiggins, 2024-10-16:
Users flagged that the new updates to DNA tracing significantly increase the amount of outputs produced when `image_set: all` is set in the config file. It was agreed that as all output images have potential to be useful for parameter refining, rather than reduce the output it would be better to have the option to set whether `core` or `all` should be output for filters, grains and tracing separately.

ns-rse, 2024-10-16:
@llwiggins @SylviaWhittle @MaxGamill-Sheffield @alicepyne 

This feels like this is something that needs addressing _before_ merging and making a release, do you agree?

Regardless, what should the defaults be with regards to setting `core` / `all` for each stage of `filters` / `grains` / `tracing` ?

alicepyne, 2024-10-16:
I think defaults to core but would say that separating these can wait until
the next smaller release to avoid scope creep in this one

SylviaWhittle, 2024-10-16:
With how urgent this is to merge (so reviewers can verify the code works) this could be done after a merge I think? It would be preferable to do beforehand aye but I believe the priority currently is that reviewers are able to run the code from main?

ns-rse, 2024-10-16:
It will be useful to set some clear Milestones for the coming months work to prioritise the backlog of issues and what needs addressing first (and clear out the existing Milestones!).

PR #1127 (2025-04-14): Selective plotting of TopoStats images
# TopoStats Pull Requests

Closes #958. This turns the config option `image_set` to use a list of strings. The strings relate to each module and then enable plotting of a few or many specified module images in order to speed-up processing on large datasets where some but not all diagnostic images are required.

This retains the single use all / core functionality but extends this to use combinations of:
- filters
- grain
- grain_crops
- disordered_tracing
- nodestats
- ordered_tracing
- splining

This includes updating the docs, adapting old tests, and cr [... cut]
closed 2025-04-14 by MaxGamill-Sheffield