AFM-SPM/TopoStats #960: [feature] : Height profiles for rotational heterogeneity

MaxGamill-Sheffield, 2024-10-16:
### Is your feature request related to a problem?

Ideally the feret height profiles are used for "blob" analysis and look for spatial heterogeneity (i.e. ndp52's coiled-coil making it flexible), but height profiles along the horizontal scan lines are should be able to identify rotational (or deposition-based) heterogeneity, and should be more stable along one scan line. One can get an idea of this heterogeneity by plotting the height profiles to see the spread of the data across all identified objects.

### Describe the solution you would like.

Add horizontal height profiles to the already obtained feret height profile with the cumulative distances in the output alongside the heights.

MaxGamill-Sheffield, 2024-10-16:
> Would this not suffer from the same issue of height profiles that the feret profile sought to resolve?
Yep, the idea is to measure two different things via the two different height profiles.

> Namely you can not guarantee the orientation of the molecules will always be in the same plane to start with (viz rotating a microscope slide on the platform you then are looking at it in a different orientation when viewed from above).
Exactly, so where the the ferets are super useful for looking how the molecules shape differs, this would be useful to see how their orientation differ - which might be due to depositional effects, and is less prone to flattening errors.

ns-rse, 2024-10-17:
Aren't there going to be tons of factors that can't be controlled that affect how molecules sit on the surface such as the angle the solution is placed on the surface, purity of buffers, electrostatic charges on surface material (van der Waals forces?) and more things I know nothing of.

Notionally its not going to be super hard to take the cropped grains and take a line across the mid-point of the rows.