AFM-SPM/TopoStats #969: [Bug]: Concatenation warning that stops all_statistics.csv from being produced

llwiggins, 2024-10-16:
### Checklist

- [X] Re-run analysis with `topostats process --core 1`.
- [X] Describe the bug.
- [X] Include the configuration file.
- [X] Copy of the output.
- [X] The exact command that failed. This is what you typed at the command line, including any options.
- [X] TopoStats version, this is reported by `topostats --version`
- [X] Operating System and Python Version

### Describe the bug

Both @MaxGamill-Sheffield and keep running into the same concatenation warning when running `topostats process`. This warning occurs right at the end of processing and results in no all_statistics.csv being output. It looks as though the issue arises from deprecation of the function that originally concatenated empty or all NA data frames, and the suggested resolution is to exclude these prior to concatenation. 

### Copy of the output

Traceback (most recent call last):
  File "/Users/laura/miniconda3/envs/topoly/bin/topostats", line 8, in <module>
    sys.exit(entry_point())
             ^^^^^^^^^^^^^
  File "/Users/laura/TopoStats/topostats/entry_point.py", line 386, in entry_point
    args.func(args)
  File "/Users/laura/TopoStats/topostats/run_topostats.py", line 171, in run_topostats
    results = pd.concat(results.values())
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/laura/miniconda3/envs/topoly/lib/python3.11/site-packages/pandas/core/reshape/concat.py", line 395, in concat
    return op.get_result()
           ^^^^^^^^^^^^^^^
  File "/Users/laura/miniconda3/envs/topoly/lib/python3.11/site-packages/pandas/core/reshape/concat.py", line 684, in get_result
    new_data = concatenate_managers(
               ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/laura/miniconda3/envs/topoly/lib/python3.11/site-packages/pandas/core/internals/concat.py", line 189, in concatenate_managers
    values = _concatenate_join_units(join_units, copy=copy)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/laura/miniconda3/envs/topoly/lib/python3.11/site-packages/pandas/core/internals/concat.py", line 491, in _concatenate_join_units
    warnings.warn(
FutureWarning: The behavior of DataFrame concatenation with empty or all-NA entries is deprecated. In a future version, this will no longer exclude empty or all-NA columns when determining the result dtypes. To retain the old behavior, exclude the relevant entries before the concat operation.

### Include the configuration file

```
base_dir: /Volumes/shared/pyne_group/Shared/AFM_Data/Metallodrugs/data # Directory in which to search for data files
output_dir: /Volumes/shared/pyne_group/Shared/AFM_Data/Metallodrugs/output # Directory to output results to
log_level: info # Verbosity of output. Options: warning, error, info, debug
cores: 1 # Number of CPU cores to utilise for processing multiple files simultaneously.
file_ext: .spm # File extension of the data files.
loading:
  channel: Height # Channel to pull data from in the data files.
filter:
  run: true # Options : true, false
  row_alignment_quantile: 0.5 # lower values may improve flattening of larger features
  threshold_method: std_dev # Options : otsu, std_dev, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    below: 10.0 # Threshold for data below the image background
    above: 1.0 # Threshold for data above the image background
  threshold_absolute:
    below: -1.0 # Threshold for data below the image background
    above: 1.0 # Threshold for data above the image background
  gaussian_size: 1.0121397464510862 # Gaussian blur intensity in px
  gaussian_mode: nearest
  # Scar remvoal parameters. Be careful with editing these as making the algorithm too sensitive may
  # result in ruining legitimate data.
  remove_scars:
    run: true
    removal_iterations: 2 # Number of times to run scar removal.
    threshold_low: 0.250 # lower values make scar removal more sensitive
    threshold_high: 0.666 # lower values make scar removal more sensitive
    max_scar_width: 4 # Maximum thichness of scars in pixels.
    min_scar_length: 16 # Minimum length of scars in pixels.
[... 70 lines cut ...]
  pixel_interpolation: null # Options : https://matplotlib.org/stable/gallery/images_contours_and_fields/interpolation_methods.html
  image_set: core # Options : all, core
  zrange: [-2, 5] # low and high height range for core images (can take [null, null]). low <= high
  colorbar: true # Options : true, false
  axes: true # Options : true, false (due to off being a bool when parsed)
  num_ticks: [null, null] # Number of ticks to have along the x and y axes. Options : null (auto) or integer > 1
  cmap: null # Colormap/colourmap to use (default is 'nanoscope' which is used if null, other options are 'afmhot', 'viridis' etc.)
  mask_cmap: blue_purple_green # Options : blu, jet_r and any in matplotlib
  histogram_log_axis: false # Options : true, false
summary_stats:
  run: true # Whether to make summary plots for output data
  config: null
```

### TopoStats Version

Git main branch

### Python Version

3.11

### Operating System

MacOS M1/M2 (post-2021)

MaxGamill-Sheffield, 2024-10-16:
Just adding that when I found this issue, I had all `dnatracing`, `plotting` and `summary_stats` turned off.

ns-rse, 2024-10-16:
Pandas version at the very least would be useful to know as its `pd.concat()` that does the concatenation.

llwiggins, 2024-10-16:
pandas v2.2.3

MaxGamill-Sheffield, 2024-10-17:
Small update where I got with this earlier using Laura's smaller test set.

- Managed to replicate on two images (must be >= two) for the dataframe concatenation mismatch). These are attached in minimal_topostats_example.zip below. [minimal_topostats_example.zip](https://github.com/user-attachments/files/17421487/minimal_example.zip)
- Fails on the concatenation of each images grainstats.csv / `result` (different to the failure of `imagestats.csv` I saw earlier).
- Saving the csv's (attatched) and writing a script (below) to concat the dataframe from the values of a dictionary, in a similar way to `run_topostats.py` causes no issue in the same environment 🙃. [saved_csvs.zip](https://github.com/user-attachments/files/17421560/saved_csvs.zip).
```
import pandas as pd
import numpy as np
from collections import defaultdict

base = "/Users/Maxgamill/Desktop/Uni/PhD/topo_test/TopoStats/concat/"
img1 = "20230526_puc19_tube1_24hr_mg.0_00003"
img2 = "20230526_puc19_tube1_24hr_mg.0_00002"

results = defaultdict()
for img in [img1, img2]:
    df = pd.read_csv(base+img+".csv")
    results[img] = df

total_df = pd.concat(results.values())
```
- I've made branch `maxgamill-sheffield/969-concat-issue` in which I've attempted to make a few fixes:
  - The `folder_<stats>.csv` was being overwritten by the dis and mol stats so that has been modified to produce all folder stats.
  - The error / failed outputs of the better tracing pipeline now add the columns that should have been added should it have succeeded.
- Thought it might have been because of columns that are present in one but not the other due to failure but alas nope.

Package list:
```
absl-py                   2.1.0                    pypi_0    pypi
accessible-pygments       0.0.5                    pypi_0    pypi
afmreader                 0.0.1                    pypi_0    pypi
alabaster                 0.7.16                   pypi_0    pypi
appnope                   0.1.4                    pypi_0    pypi
argparse                  1.4.0                    pypi_0    pypi
astroid                   3.1.0                    pypi_0    pypi
asttokens                 2.4.1                    pypi_0    pypi
astunparse                1.6.3                    pypi_0    pypi
babel                     2.16.0                   pypi_0    pypi
backcall                  0.2.0                    pypi_0    pypi
beautifulsoup4            4.12.3                   pypi_0    pypi
biopython                 1.84                     pypi_0    pypi
black                     24.4.2                   pypi_0    pypi
bzip2                     1.0.8                h93a5062_5    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
certifi                   2024.7.4                 pypi_0    pypi
cfgv                      3.4.0                    pypi_0    pypi
charset-normalizer        3.3.2                    pypi_0    pypi
cheap-repr                0.5.1                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
cloudpickle               3.0.0                    pypi_0    pypi
comm                      0.2.2                    pypi_0    pypi
contourpy                 1.2.1                    pypi_0    pypi
coverage                  7.5.1                    pypi_0    pypi
cycler                    0.12.1                   pypi_0    pypi
debugpy                   1.8.1                    pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
dill                      0.3.8                    pypi_0    pypi
distlib                   0.3.8                    pypi_0    pypi
docutils                  0.20.1                   pypi_0    pypi
entrypoints               0.4                      pypi_0    pypi
et-xmlfile                1.1.0                    pypi_0    pypi
exceptiongroup            1.2.1                    pypi_0    pypi
execnet                   2.1.1                    pypi_0    pypi
executing                 2.0.1                    pypi_0    pypi
filelock                  3.14.0                   pypi_0    pypi
filetype                  1.2.0                    pypi_0    pypi
flatbuffers               24.3.25                  pypi_0    pypi
fonttools                 4.51.0                   pypi_0    pypi
gast                      0.6.0                    pypi_0    pypi
google-pasta              0.2.0                    pypi_0    pypi
[... 126 lines cut ...]
termcolor                 2.4.0                    pypi_0    pypi
threadpoolctl             3.5.0                    pypi_0    pypi
tifffile                  2024.5.3                 pypi_0    pypi
tk                        8.6.13               h5083fa2_1    conda-forge
tokenize-rt               5.2.0                    pypi_0    pypi
tomli                     2.0.1                    pypi_0    pypi
tomlkit                   0.12.5                   pypi_0    pypi
toolz                     0.12.1                   pypi_0    pypi
topofileformats           0.1.0                    pypi_0    pypi
topoly                    1.0.2                    pypi_0    pypi
topostats                 2.2.2.dev896+gcc66a1fa9          pypi_0    pypi
tornado                   6.4                      pypi_0    pypi
tqdm                      4.66.4                   pypi_0    pypi
traitlets                 5.14.3                   pypi_0    pypi
typing-extensions         4.11.0                   pypi_0    pypi
[... cut ...]
urllib3                   2.2.2                    pypi_0    pypi
virtualenv                20.26.1                  pypi_0    pypi
wcwidth                   0.2.13                   pypi_0    pypi
werkzeug                  3.0.4                    pypi_0    pypi
wheel                     0.43.0             pyhd8ed1ab_1    conda-forge
wrapt                     1.16.0                   pypi_0    pypi
wurlitzer                 3.1.0                    pypi_0    pypi
xz                        5.2.6                h57fd34a_0    conda-forge
```

ns-rse, 2024-10-21:
Re-opening as #973 is still open.

ns-rse, 2024-10-24:
Closed by #973

PR #972 (2024-10-18): Fixes `pd.concat` error halting processing
closes #969

TLDR:
PR adds the following:
- Rejigging tracing stats outputs so base grainstats / molstats / None is returned.
- Enables folder stats to be output properly.
- drops empty stats columns causing concat warning.

Think I got it. The cause according to the docs is that pandas doesn't like concatenating objects with whole columns (or rows) as NaN's or None's (we had a mix which is why it was hard to find).

Not only that, but for some reason the "warning" also halts processing.

The initial fix I tried was to stop the warning using the below [... cut]
closed 2024-10-24 by ns-rse