AFM-SPM/TopoStats #1021: Expand CITATION.cff

ns-rse, 2024-11-20:
The `CITATION.cff` is missing details of earlier contributors to the package such as those listed in [TopoStats – A program for automated tracing of biomolecules from AFM images - ScienceDirect](https://www.sciencedirect.com/science/article/pii/S1046202321000207).

We should add these authors to the `CITATION.cff`, ideally ensuring we have their current contact details and metadata...

- `email`
- `affiliation`
- `orcid`

At the very least if people can't be contacted we should be able to find and add their [ORCiD](https://orcid.org).

ns-rse, 2024-11-25:
Reading through the [paper](https://www.sciencedirect.com/science/article/pii/S1046202321000207) and the _CRediT authorship contribution statement_ section those noted to have contributed to the software for whom I don't have contact details are...

```
  - given-names: Joseph
    family-names: Beton
    email:
    affiliation:
    orcid:
  - given-names: Luzie
    family-names: Helfmann
    email:
    affiliation:
    orcid:
  - given-names: Robert
    family-names: Gray
    email:
    affiliation:
    orcid:
  - given-names: Agnel Praveen
    family-names: Joseph
    email:
    affiliation:
    orcid:
```

alicepyne, 2024-11-26:
https://www.scd.stfc.ac.uk/Pages/Agnel-Praveen-Joseph.aspx

commit 7c017af9 (2024-11-27): chore: Updates publication in README.md

Closes #1021
PR #1023 (2024-11-27): chore: Adds earlier contributors to CITATION.cff
Closes #1021

If anyone has missing information (mostly email and one ORCID) for those being added please leave as a suggestion in the Pull Request and we can add it in.

---

Before submitting a Pull Request please check the following.

**NA** No code changes in this PR

- [ ] ~~Existing tests pass.~~
- [ ] ~~Documentation has been updated and builds. Remember to update as required...~~
  - [ ] ~~`docs/configuration.md`~~
  - [ ] ~~`docs/usage.md`~~
  - [ ] ~~`docs/data_dictionary.md`~~
  - [ ] ~~`docs/advanced.md` and new pages it should link  [... cut]
commit bce42c63 (2024-11-27): chore: Adds earlier contributors to CITATION.cff

Closes #1021

If anyone has missing information (mostly email and one ORCID) for those being added please leave as a suggestion in the
Pull Request and we can add it in.
closed 2024-11-27 by ns-rse