AFM-SPM/TopoStats #1037: [Bug]: Possibly need to pin AFMReader version?

SylviaWhittle, 2024-12-05:
### Checklist

- [X] Re-run analysis with `topostats process --core 1`.
- [X] Describe the bug.
- [X] Include the configuration file.
- [X] Copy of the output.
- [X] The exact command that failed. This is what you typed at the command line, including any options.
- [X] TopoStats version, this is reported by `topostats --version`
- [X] Operating System and Python Version

### Describe the bug

TLDR:
- Tobi found a bug where AFMReader couldn't load a particular `.spm` file due to units that were present in TopoStats not being present in AFMReader.
- This was solved with AFMReader release `v0.0.3`
- Should we thus require AFMReader >= `v0.0.3` in TopoStats?
- I have also had two users ask me how to fix the error where TopoStats can't import `gwy` from AFMReader (since they were using an old AFMReader version), the solution was to `pip install --upgrade AFMReader`, if we required AFMReader >= `v0.0.3` then this wouldn't happen?

### Copy of the output

Using cached argparse-1.4.0-py2.py3-none-any.whl (23 kB)
Building wheels for collected packages: topostats
  Building editable for topostats (pyproject.toml) ... done
  Created wheel for topostats: filename=topostats-2.2.2.dev1258+g85d65c33-0.editable-py3-none-any.whl size=21309 sha256=c66a320fffb74a8c02b6afafe81d1adc55cff8cb488959496318814b744a7cd7
  Stored in directory: /private/var/folders/tm/prjlcf551xq94z0v8y336h5r0000gn/T/pip-ephem-wheel-cache-9rnj70a4/wheels/1b/4f/34/8f66d079523245e660d0636b8079af9b3510b3c8818abf4e9d
Successfully built topostats
Installing collected packages: argparse, topostats
  Attempting uninstall: topostats
    Found existing installation: topostats 2.2.2.dev1218+gc8dc6619
    Uninstalling topostats-2.2.2.dev1218+gc8dc6619:
      Successfully uninstalled topostats-2.2.2.dev1218+gc8dc6619
Successfully installed argparse-1.4.0 topostats-2.2.2.dev1258+g85d65c33
(main_topostats) t@dyn232037 TopoStats % topostats process
Traceback (most recent call last):
  File "/Users/t/miniconda3/envs/main_topostats/bin/topostats", line 5, in <module>
    from topostats.entry_point import entry_point
  File "/Users/t/Documents/topostats/main_topostats/TopoStats/topostats/entry_point.py", line 11, in <module>
    from topostats import __version__, run_modules
  File "/Users/t/Documents/topostats/main_topostats/TopoStats/topostats/run_modules.py", line 22, in <module>
    from topostats.io import (
  File "/Users/t/Documents/topostats/main_topostats/TopoStats/topostats/io.py", line 22, in <module>
    from AFMReader import asd, gwy, ibw, jpk, spm, topostats
ImportError: cannot import name 'gwy' from 'AFMReader' (/Users/t/miniconda3/envs/main_topostats/lib/python3.10/site-packages/AFMReader/__init__.py)
(main_topostats) t@dyn232037 TopoStats % cd "/Users/t/Documents/430 minicircles topostats test for Niel"
(main_topostats) t@dyn232037 430 minicircles topostats test for Niel % topostats create-config --filename config.yaml
Traceback (most recent call last):
  File "/Users/t/miniconda3/envs/main_topostats/bin/topostats", line 5, in <module>
    from topostats.entry_point import entry_point
  File "/Users/t/Documents/topostats/main_topostats/TopoStats/topostats/entry_point.py", line 11, in <module>
    from topostats import __version__, run_modules
  File "/Users/t/Documents/topostats/main_topostats/TopoStats/topostats/run_modules.py", line 22, in <module>
    from topostats.io import (
  File "/Users/t/Documents/topostats/main_topostats/TopoStats/topostats/io.py", line 22, in <module>
    from AFMReader import asd, gwy, ibw, jpk, spm, topostats
ImportError: cannot import name 'gwy' from 'AFMReader' (/Users/t/miniconda3/envs/main_topostats/lib/python3.10/site-packages/AFMReader/__init__.py)
(main_topostats) t@dyn232037 430 minicircles topostats test for Niel % topostats create-config --filename config.yaml                 
Traceback (most recent call last):
  File "/Users/t/miniconda3/envs/main_topostats/bin/topostats", line 5, in <module>
    from topostats.entry_point import entry_point
  File "/Users/t/Documents/topostats/main_topostats/TopoStats/topostats/entry_point.py", line 11, in <module>
    from topostats import __version__, run_modules
  File "/Users/t/Documents/topostats/main_topostats/TopoStats/topostats/run_modules.py", line 22, in <module>
    from topostats.io import (
  File "/Users/t/Documents/topostats/main_topostats/TopoStats/topostats/io.py", line 22, in <module>
    from AFMReader import asd, gwy, ibw, jpk, spm, topostats
ImportError: cannot import name 'gwy' from 'AFMReader' (/Users/t/miniconda3/envs/main_topostats/lib/python3.10/site-packages/AFMReader/__init__.py)

### Include the configuration file

N/A: Any will trigger this provided using `.spm` file format and the correct file (ask Tobi) if reproducing the units bug in AFMReader (requires AFMReader < `v0.0.3`)

### To Reproduce

Run TopoStats with `AFMReader` < `v0.0.2` for the gwy bug, or < `v0.0.3` for the units bug.

### TopoStats Version

Git main branch

### Python Version

3.11

### Operating System

MacOS M1/M2 (post-2021)

### Python Packages

AFMReader < `v0.0.3` or < `v0.0.2`
Else: any

ns-rse, 2024-12-10:
I've just deleted AFMReader v0.0.2 from PyPI so it should now be impossible for anyone to install that broken release.

This negates the need to pin the version of AFMReader that is installed.

Given `v0.0.2` was released 2024-12-04 and the problem was fixed the following day with `v0.0.3` released 2024-12-05 I don't think we're going to encounter many people who hit this problem and we perhaps don't need to pin the dependency.

Let me know your thoughts @SylviaWhittle

ns-rse, 2024-12-11:
This will have affected very few people, the users have all been advised to upgrade AFMReader to `v0.0.3` and we have removed `v0.0.2` from PyPI so no one will be affected in the future.

closed 2024-12-11 by ns-rse