AFM-SPM/TopoStats #1066: [Bug]: Bounding boxes for grain masks seem to be disrupting other masks and traces

tcatley, 2025-01-03:
### Checklist

- [X] Find the offending file in the output. If processing halts, re-run analysis with `topostats --core 1 process`.
- [X] Describe the bug.
- [X] Include the configuration file.
- [X] Copy of the log-file from running with `topostats --log-level debug <command>`.
- [X] The exact command that failed. This is what you typed at the command line, including any options.
- [X] TopoStats version, this is reported by `topostats --version`
- [X] Operating System and Python Version

### Describe the bug

I have an images with lots of overlapping molecules. The new bounding boxes which are drawn around the masks seem to be chopping up the masks on adjacent grains and effecting the subsequent traces. 
[image]
The centre bottom and the middle right of the mask image shows masks which have been cut off by the corner of a nearby mask. The spline and trace images show that this cutoff has also disrupted the traces. 
[image]
[image]

### Copy of the log-file from running with `topostats --log-level debug <command>`

[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [validation.py] [32] The YAML configuration file is valid.
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [validation.py] [32] The YAML plotting configuration file is valid.
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [processing.py] [1316] Configuration run options are consistent, processing can proceed.
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [run_modules.py] [139] Configuration file loaded from      : /Users/work/Documents/new_config.yaml
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [run_modules.py] [140] Scanning for images in              : /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [run_modules.py] [141] Output directory                    : /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [run_modules.py] [142] Looking for images with extension   : .spm
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [run_modules.py] [143] Images with extension .spm in /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure : 4
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [run_modules.py] [148] Thresholding method (Filtering)     : std_dev
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [run_modules.py] [149] Thresholding method (Grains)        : absolute
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [io.py] [748] Extracting image from /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/20240802_GMP_RTdryup_KCl2.0_00017.spm
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [io.py] [748] Extracting image from /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/20240802_GMP_RTdryup_KCl2.0_00014.spm
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [io.py] [748] Extracting image from /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/Extra images same sample/20240802_GMP_RTdryup_KCl2.0_00016.spm
[Fri, 03 Jan 2025 10:22:42] [INFO    ] [topostats] [io.py] [748] Extracting image from /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/Extra images same sample/20240802_GMP_RTdryup_KCl2.0_00021.spm
[Fri, 03 Jan 2025 10:23:09] [INFO    ] [topostats] [run_modules.py] [275] [20240802_GMP_RTdryup_KCl2.0_00017] Processing completed.
[Fri, 03 Jan 2025 10:23:20] [INFO    ] [topostats] [run_modules.py] [275] [20240802_GMP_RTdryup_KCl2.0_00021] Processing completed.
[Fri, 03 Jan 2025 10:23:23] [INFO    ] [topostats] [run_modules.py] [275] [20240802_GMP_RTdryup_KCl2.0_00014] Processing completed.
[Fri, 03 Jan 2025 10:23:30] [INFO    ] [topostats] [run_modules.py] [275] [20240802_GMP_RTdryup_KCl2.0_00016] Processing completed.
[Fri, 03 Jan 2025 10:23:30] [INFO    ] [topostats] [run_modules.py] [277] Saving image stats to : /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/image_stats.csv.
[Fri, 03 Jan 2025 10:23:30] [INFO    ] [topostats] [run_modules.py] [302] Saving all height profiles to /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/height_profiles.json
[Fri, 03 Jan 2025 10:23:30] [INFO    ] [topostats] [io.py] [397] No folder-wise statistics for directory nan, no grains detected in any images.
[Fri, 03 Jan 2025 10:23:30] [INFO    ] [topostats] [io.py] [397] No folder-wise statistics for directory nan, no grains detected in any images.
[Fri, 03 Jan 2025 10:23:30] [INFO    ] [topostats] [io.py] [395] Folder-wise statistics saved to: /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/folder_grain_stats.csv
[Fri, 03 Jan 2025 10:23:30] [INFO    ] [topostats] [io.py] [395] Folder-wise statistics saved to: /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/Extra images same sample/folder_grain_stats.csv
[Fri, 03 Jan 2025 10:23:30] [INFO    ] [topostats] [io.py] [395] Folder-wise statistics saved to: /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/folder_disordered_trace_stats.csv
[Fri, 03 Jan 2025 10:23:30] [INFO    ] [topostats] [io.py] [395] Folder-wise statistics saved to: /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/Extra images same sample/folder_disordered_trace_stats.csv
[Fri, 03 Jan 2025 10:23:31] [INFO    ] [topostats] [io.py] [395] Folder-wise statistics saved to: /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/folder_mol_stats.csv
[Fri, 03 Jan 2025 10:23:31] [INFO    ] [topostats] [io.py] [395] Folder-wise statistics saved to: /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/Extra images same sample/folder_mol_stats.csv
[Fri, 03 Jan 2025 10:23:31] [INFO    ] [topostats] [processing.py] [1389] 

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ COMPLETE ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

  TopoStats Version           : 2.2.2.dev1320+gcf4682483
  Base Directory              : /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure
  File Extension              : .spm
  Files Found                 : 4
  Successfully Processed^1    : 4 (100.0%)
  All statistics              : /Users/work/Library/CloudStorage/GoogleDrive-t.catley@sheffield.ac.uk/Shared drives/Pyne_group_data/Papers/Simon Quadruplex manuscript/TopoStats data analysis (sharred)/Figure_2_GQs_structure/all_statistics.csv
  Distribution Plots          : Disabled. Enable in config 'summary_stats/run' if needed.

[... cut ...]
  style: topostats.mplstyle
  zrange: [-3, 4]
splining: {method: rolling_window, rolling_window_size: 2e-08, run: true, spline_circular_smoothing: 5.0,
  spline_degree: 3, spline_linear_smoothing: 5.0, spline_step_size: 7e-09}
summary_stats: {config: null, run: false}

### To Reproduce

I can provide the file on request - it is for an external collaborator paper so don't want to post here. 

### TopoStats Version

Git main branch

### Python Version

3.11

### Operating System

MacOS M1/M2 (post-2021)

MaxGamill-Sheffield, 2025-01-08:
Seems to be an issue made in #922 the as #988 (I believe this to be the previous PR) seems to work alright. See pictures below.

[image]

Currently debating wether this needs to be fixed ASAP or if it will be resolved in the upcoming grains refactor.

ns-rse, 2025-01-09:
Why would enabling tests in Continuous Integration or configuration files cause problems in plotting? The changes in the configuration file were all to comments too.

How did you narrow this down? Was it using `git bisect` to find the offending commit?

MaxGamill-Sheffield, 2025-01-09:
> Why would enabling tests in Continuous Integration or configuration files cause problems in plotting? The changes in the configuration file were all to comments too.
Apologies I meant 992, I've corrected this now.

> How did you narrow this down? Was it using `git bisect` to find the offending commit?
And I found this when running the last commit of this vs the last commit of PR prior to 922. What does git bisect do?

PR #1070 (2025-01-09): Combine grain crops rather than overwriting them
Resolves #1066 

This PR resolves the bug found by @tcatley in issue #1066 by combining grain crops rather than overwriting them to avoid issues when multiple bounding boxes overlap one another. 

The image below shows how grain crops are now correctly combined to assemble full image masks.

[image]
closed 2025-01-09 by llwiggins