AFM-SPM/TopoStats #1081: [Bug]: Get an error saving the height profiles when saving grains but not tracing (`disordered_tracing` set to `false`)

derollins, 2025-01-28:
### Checklist

- [x] Find the offending file in the output. If processing halts, re-run analysis with `topostats --core 1 process`.
- [x] Describe the bug.
- [x] Include the configuration file.
- [x] Copy of the log-file from running with `topostats --log-level debug <command>`.
- [x] The exact command that failed. This is what you typed at the command line, including any options.
- [x] TopoStats version, this is reported by `topostats --version`
- [x] Operating System and Python Version

### Describe the bug

Hello, I'm back! With another error!

When `grainstats` is set to `true` while `disordered_traces` is set to `false` I get this error:

`[Tue, 28 Jan 2025 16:58:07] [INFO    ] [topostats] Saving all height profiles to output2b/height_profiles.json
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\ggjh246\AppData\Local\anaconda3\envs\topostats_git2\Scripts\topostats.exe\__main__.py", line 7, in <module>
  File "C:\Users\ggjh246\Code\TopoStats\topostats\entry_point.py", line 1214, in entry_point
    args.func(args)
  File "C:\Users\ggjh246\Code\TopoStats\topostats\run_modules.py", line 361, in process
    results.set_index(["image", "threshold", "grain_number"], inplace=True)
  File "C:\Users\ggjh246\AppData\Local\anaconda3\envs\topostats_git2\Lib\site-packages\pandas\core\frame.py", line 6122, in set_index
    raise KeyError(f"None of {missing} are in the columns")
KeyError: "None of ['grain_number'] are in the columns"
Exception ignored in: <_io.FileIO name='C:\\Users\\ggjh246\\test_data\\save-2025.01.22-14.42.45.705.jpk' mode='rb' closefd=True>
ResourceWarning: unclosed file <_io.BufferedReader name='C:\\Users\\ggjh246\\test_data\\save-2025.01.22-14.42.45.705.jpk'>
Exception ignored in: <_io.FileIO name='C:\\Users\\ggjh246\\test_data\\save-2025.01.22-14.43.02.844.jpk' mode='rb' closefd=True>
ResourceWarning: unclosed file <_io.BufferedReader name='C:\\Users\\ggjh246\\test_data\\save-2025.01.22-14.43.02.844.jpk'>`

This prevents running the grain analysis without tracing, slowing topostats down when tracing is not desired and generating unwated data.

### Copy of the log-file from running with `topostats --log-level debug <command>`

[topostatserrror.txt](https://github.com/user-attachments/files/18577334/topostatserrror.txt)

```[Tue, 28 Jan 2025 17:11:53] [INFO    ] [topostats] [save-2025.01.22-14.43.07.129] : Calculation of nodestats disabled, returning empty dataframe.
[Tue, 28 Jan 2025 17:11:53] [INFO    ] [topostats] [save-2025.01.22-14.43.07.129] : Calculation of Curvature Stats disabled, returning None.
[Tue, 28 Jan 2025 17:11:53] [INFO    ] [topostats] [save-2025.01.22-14.43.07.129] : *** Image Statistics ***
[Tue, 28 Jan 2025 17:11:53] [INFO    ] [topostats] [save-2025.01.22-14.43.07.129] : Saving image to .topostats file
Processing images from C:\Users\ggjh246\test_data, results are under output2b: 100%|█████| 8/8 [02:21<00:00, 16.76s/it][Tue, 28 Jan 2025 17:11:53] [INFO    ] [topostats] [save-2025.01.22-14.43.07.129] Processing completed.
Processing images from C:\Users\ggjh246\test_data, results are under output2b: 100%|█████| 8/8 [02:21<00:00, 17.70s/it]
[Tue, 28 Jan 2025 17:11:53] [INFO    ] [topostats] Saving image stats to : output2b/image_stats.csv.
[Tue, 28 Jan 2025 17:11:53] [INFO    ] [topostats] Saving all height profiles to output2b/height_profiles.json
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\ggjh246\AppData\Local\anaconda3\envs\topostats_git2\Scripts\topostats.exe\__main__.py", line 7, in <module>
  File "C:\Users\ggjh246\Code\TopoStats\topostats\entry_point.py", line 1214, in entry_point
    args.func(args)
  File "C:\Users\ggjh246\Code\TopoStats\topostats\run_modules.py", line 361, in process
    results.set_index(["image", "threshold", "grain_number"], inplace=True)
  File "C:\Users\ggjh246\AppData\Local\anaconda3\envs\topostats_git2\Lib\site-packages\pandas\core\frame.py", line 6122, in set_index
    raise KeyError(f"None of {missing} are in the columns")
KeyError: "None of ['grain_number'] are in the columns"
Exception ignored in: <_io.FileIO name='C:\\Users\\ggjh246\\test_data\\save-2025.01.22-14.42.45.705.jpk' mode='rb' closefd=True>
ResourceWarning: unclosed file <_io.BufferedReader name='C:\\Users\\ggjh246\\test_data\\save-2025.01.22-14.42.45.705.jpk'>
Exception ignored in: <_io.FileIO name='C:\\Users\\ggjh246\\test_data\\save-2025.01.22-14.43.02.844.jpk' mode='rb' closefd=True>
ResourceWarning: unclosed file <_io.BufferedReader name='C:\\Users\\ggjh246\\test_data\\save-2025.01.22-14.43.02.844.jpk'>
```
Rest of file is in file.

### Include the configuration file

```
# Config file generated 2025-01-28 14:44:34
# # For more information on configuration and how to use it:
[... 219 lines cut ...]
six==1.17.0
skan==0.12.2
snoop==0.6.0
stack-data==0.6.3
superqt==0.7.1
tensorboard==2.18.0
tensorboard-data-server==0.7.2
tensorflow==2.18.0
tensorflow-io-gcs-filesystem==0.31.0
tensorflow_intel==2.18.0
termcolor==2.5.0
threadpoolctl==3.5.0
tifffile==2025.1.10
toolz==1.0.0
topoly==1.0.4
-e git+https://github.com/AFM-SPM/TopoStats.git@c769ca3cfd32bf37ab3a158cf032b80fa5bcf51a#egg=topostats
tqdm==4.67.1
traitlets==5.14.3
typing_extensions==4.12.2
tzdata==2025.1
urllib3==2.3.0
wcwidth==0.2.13
Werkzeug==3.1.3
win32_setctime==1.2.0
wrapt==1.17.2

MaxGamill-Sheffield, 2025-02-25:
Hey @derollins!

I believe this is caused by the way that some of the results files are indexed - by grain number instead of by index with a column of grain number.

@SylviaWhittle has found this issue and fixed it in her upcoming refactor so keep an eye out for that! In the meantime can I recommend you try including running disordered tracing (using Zhang skeletonisation)? This shouldn't add too much time onto processing but should jiggle around the grainstats columns.

Please lmk if that's worked for you :)

derollins, 2025-02-27:
Thanks, that works although while if I plot `all` the grain images are numbered, there are no numbers in `all_ststistics.csv` for grains that weren't sucessfully traced. Is this from the same issue?