AFM-SPM/TopoStats #1093: Add the ability to disable logging

ns-rse, 2025-02-21:
@tdjames1 highlighted that all of the modules include logging such that when used in isolation they produce lots of
`.log` files.

The entry-point has the ability to run `--quiet` which reduces output and #929 helped considerably.

It would however be useful to add a `--silent` argument to suppress all output and ensure that individual modules don't
perform logging when used in isolation.

ns-rse, 2025-02-24:
Hi @tdjames,

I want to test that what I think is a solution works for your use case. Do you have a code snippet you could share to show how you are using TopoStats please?

tdjames1, 2025-02-25:
Here's a snippet showing the noisy section:
```
from topostats.io import LoadScans

img_files = [  ... list of paths to .spm files ... ]
load_config = {'channel': 'Height'}

# Load scans
loadscans = LoadScans(img_files, **load_config)
loadscans.get_data()
```

tdjames1, 2025-02-25:
I've also got calls which use `Filters.filter_image()` which I think also created a lot of output, but I'm not triggering those atm.