AFM-SPM/TopoStats #1110: AFMReader JPK changes break TopoStats tests

ns-rse, 2025-03-18:
In merging [AFMReader #118](https://github.com/AFM-SPM/AFMReader/pull/118) it appears to have broken the tests in
TopoStats. Whilst files load the summation of the array differs as seen in [this CI
run](https://github.com/AFM-SPM/TopoStats/actions/runs/13921465417/job/38955518431?pr=1105)

``` shell
=========================== short test summary info ============================
666
FAILED tests/test_io.py::test_load_scan_jpk - assert 219242202.8256843 == 286598232.9308627
667
 +  where 219242202.8256843 = <built-in method sum of numpy.ndarray object at 0x140bb01b0>()
668
 +    where <built-in method sum of numpy.ndarray object at 0x140bb01b0> = array([[3334.48515103, 3334.28203109, 3334.85736085, ..., 3315.56341253,\n        3315.32033382, 3314.87393257],\n       [3334.07487533, 3333.9557643 , 3334.11107844, ..., 3315.45207676,\n        3315.38176601, 3315.1798964 ],\n       [3334.40098699, 3334.40282429, 3334.40151028, ..., 3315.95224839,\n        3315.42639502, 3315.23120084],\n       ...,\n       [3303.75336692, 3304.30197183, 3304.39811946, ..., 3353.236224  ,\n        3353.12335656, 3352.78876923],\n       [3303.52794694, 3303.81660283, 3304.39202637, ..., 3353.1202504 ,\n        3352.6498461 , 3352.59804501],\n       [3304.32763273, 3304.81272967, 3305.07541323, ..., 3353.35865769,\n        3353.2311856 , 3353.64670578]]).sum
669
!!!!!!!!!!!!!!!!!!!!!!!!!! stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!!!!
670
================== 1 failed, 380 passed, 2 warnings in 10.96s ==================
```

Need to investigate the impact of this more widely.

For the time being I have pinned the version of AFMReader that is installed (from Git rather than pip) in #1105 to
[1f03159](https://github.com/AFM-SPM/AFMReader/commit/1f03159edae2e906cb709ad9949d7b6d460cd1a9).

ns-rse, 2025-03-18:
Investigating the problem here...

``` shell
mkvirtualenv afmreader-broken
cd ~/work/git/hub/AFM-SPM/TopoStats/
git switch main
git pull
pip install -e .[dev,test]
pytest tests/test_io.py
=============================================== short test summary info ===============================================
SKIPPED [1] tests/test_io.py:549: UnboundLocalError from AFMReader.ibw.ibw_load() if file does not exist means image is None and can not be returned.
FAILED tests/test_io.py::test_load_scan_jpk - assert equals failed
FAILED tests/test_io.py::test_load_scan_get_data[jpk] - assert equals failed
======================================= 2 failed, 96 passed, 1 skipped in 6.20s =======================================
```

Checking the individual tests.

## `test_load_scan_jpk`

``` python
def test_load_scan_jpk(load_scan_jpk: LoadScans) -> None:
    """Test loading of JPK Instruments .jpk file."""
    load_scan_jpk.img_path = load_scan_jpk.img_paths[0]
    load_scan_jpk.filename = load_scan_jpk.img_paths[0].stem
    image, px_to_nm_scaling = load_scan_jpk.load_jpk()
    assert isinstance(image, np.ndarray)
    assert image.shape == (256, 256)
    assert image.sum() == 286598232.9308627
    assert isinstance(px_to_nm_scaling, float)
    assert px_to_nm_scaling == 1.2770176335964876
```

``` shell
============================================================= FAILURES ==============================================================
________________________________________________________ test_load_scan_jpk _________________________________________________________

load_scan_jpk = <topostats.io.LoadScans object at 0x76dcb5db6690>

    def test_load_scan_jpk(load_scan_jpk: LoadScans) -> None:
        """Test loading of JPK Instruments .jpk file."""
        load_scan_jpk.img_path = load_scan_jpk.img_paths[0]
        load_scan_jpk.filename = load_scan_jpk.img_paths[0].stem
        image, px_to_nm_scaling = load_scan_jpk.load_jpk()
        assert isinstance(image, np.ndarray)
        assert image.shape == (256, 256)
>       assert image.sum() == 286598232.9308627
E       assert equals failed
E         219242202.8256843  286598232.9308627

tests/test_io.py:532: AssertionError
------------------------------------------------------- Captured stderr call --------------------------------------------------------
```

## `test_load_scan_get_data[jpk]`

This is in essence the same test and unsurprisingly fails with the same error, the `image_sum` differs...

``` python
@pytest.mark.parametrize(
    ("load_scan_object", "length", "image_shape", "image_sum", "filename", "pixel_to_nm_scaling"),
    [
        pytest.param("load_scan_jpk", 1, (256, 256), 286598232.9308627, "file", 1.2770176335964876, id="jpk"),
    ],
)
def test_load_scan_get_data(
    load_scan_object: LoadScans,
    length: int,
    image_shape: tuple,
    image_sum: float,
[... 112 lines cut ...]
array([[1024.42722853, 1024.36482562, 1024.54157958, ..., 1018.61405401,
        1018.53937488, 1018.40223059],
       [1024.30118281, 1024.26458928, 1024.3123052 , ..., 1018.57984921,
        1018.5582482 , 1018.49622941],
       [1024.40137149, 1024.40193595, 1024.40153225, ..., 1018.73351294,
        1018.57195921, 1018.51199126],
       ...,
       [1014.98574806, 1015.15429157, 1015.18383023, ..., 1030.18797085,
        1030.15329549, 1030.05050289],
       [1014.91649409, 1015.00517553, 1015.1819583 , ..., 1030.15234121,
        1030.00782265, 1029.99190822],
       [1015.16217516, 1015.31120762, 1015.39190979, ..., 1030.22558517,
        1030.18642294, 1030.31407989]])

```

Comparing these images with [difftastic](https://difftastic.wilfred.me.uk/) shows that for all intent and purposes they
are the same, and so the _relative_ difference in heights between each point are identical.

``` shell
❱ difft test_jpk_plot_broken_20250318.png test_jpk_plot_original_20250318.png
test_jpk_plot_original_20250318.png --- Binary
No changes.

```

derollins, 2025-03-18:
I think this is what you expect when you change the way the images are scaled (essentially what the AFMReader fix was). I had to update similar tests in AFMReader itself and did a similar thing. I used three other programmes that open jpk files (Gwyddion, JPK data processing and Nanolocz) to verify the maximum and minimum values, and sum of the points to check the new test was for a correctly opened image.

ns-rse, 2025-03-19:
Thanks @derollins if I'd actually taken the time to go back and look at your PR in detail I'd have seen the [test file](https://github.com/AFM-SPM/AFMReader/pull/118/files) updates this sum to the same value observed here and I could have saved myself some time!

Another lesson learned and an incoming PR which fixes this.

PR #1111 (2025-03-19): tests(jpk): Updates sum of images loaded from JPK
Closes #1110

In light of the changes to AFMReader which modified the scaling the sum of the arrays of the `.jpk` test files needed
updating here too.

Before submitting a Pull Request please check the following.

- [x] Existing tests pass.
- [x] Pre-commit checks pass.
commit a1b0570c (2025-03-19): tests(jpk): Updates sum of images loaded from JPK

Closes #1110

In light of the changes to AFMReader which modified the scaling the sum of the arrays of the `.jpk` test files needed
updating here too.
closed 2025-03-19 by ns-rse