AFM-SPM/TopoStats #1113: `ImageGrainCrops` - `disorderedTrace`

ns-rse, 2025-03-27:
Refactor `disorderedTrace` class and processor functions to work with `ImageGrainCrops`.

I think we want to add all the arrays that are returned as a dictionary to be components of the `GrainCrop` class for each grain and for now also return the dictionary to retain functionality with subsequent steps. Once subsequent steps have been adapted to work with `ImageGrainCrops` (which contains nested `GrainCrop` objects/classes) we can then remove returning the dictionary.

ns-rse, 2025-04-02:
Further to comments/table in #1102 (which actually documents what `processing.run_disordered_tracing()` returns) the objects returned by `tracing.disordered_tracing.trace_image_disordered()` are...

- `disordered_trace_crop_data` - Dictionary of Numpy arrays of the disordered trace co-ordinates.
- `grainstats_additions_df` - Additional statistitics to be added to Pandas DataFrame.
- `all_images` - a dictionary of different features of the `full_image`.
- `disordered_tracing_stats` - statistics on the disordered traces.

## `disordered_trace_crop_data`

This is indexed by `grain_number` and contains a nested dictionary and contains `disordered_trace_images` (line 367)
which is the object that contains the results returned by `disordered_trace_grain` with key/values (see line
659-667 of `disordred_tracing.disordered_trace_grain()`) for...

| Key               | Value                             | Source                               |
| `cropped_image`   | `grain_crop_image`                | `grain_crop.image`                   |
| `cropped_mask`    | `grain_crop_class_mask`           | `grain_crop.mask[:,:, class_index]`  |
| `smoothed_grain`  | `disorderedtrace.smoothed_mask`   | `disorderedTrace()`                  |
| `skeleton`        | `disorderedtrace.skeleton`        | `disorceredTrace()`                  |
| `pruned_skeleton` | `disorderedtrace.pruned_skeleton` | `disorceredTrace()`                  |
| `branch_types`    |                                   | `get_skan_image(..., "branch_type")` |
| `branch_indexes`  |                                   | `get_skan_image(..., "node_id_src")` |

Each of these will need adding as an attribute to `GrainCrop` along with getter/setter methods.

## `grainstats_additions_df`

This is made from the `grainstats_additions` dictionary and so we can append/combine it to the existing
`GrainCrop.stats` attribute rather than adding a new attribute to `GrainCrop`.

## `all_images`

This is itself a dictionary with keys

- `smoothed_grain`
- `skeleton`
- `pruned_skeleton`
- `branch_indexes`
- `branch_types`

Is this duplicationn of `disordered_trace_crop_data`? Will need to check.

## `disordered_tracing_stats`

A Pandas DataFrame built from concatenating the `skan_df` dataframe initially this is the result of `skan.summarize()`
but its then modified by `compile_skan_stats()` function.

We can probably pull these out and add them to `GrainCrop.stats` too, but I'll have to check what is used from
`skan.summarize()` and what cleaning `compile_skan_stats()` does.

ns-rse, 2025-04-02:
Hi @SylviaWhittle,

I've started to look at updating aspects of `GrainCrop` class with the things
`topostats.tracing.disordered_tracing.trace_image_disordered()` produces.

First port of call is the additional statistics.

As #1118 adds `GrainCrop.stats` as an attribute I figured the simplest approach here was to merge the existing
dictionary that is stored here with the `grainstats_additions[grain_number]` that is calculated as each grain is
iterated over within the `trace_image_disordered()` function and added a line to do so...

``` python
grain_crop.stats = grain_crop.stats | grainstats_additions[grain_number]
```

Unsurprisingly and as would be expected the test `tests/tracing/test_disordered_tracing.py::test_trace_image_disordered`
now fails with...

``` python
ERROR    topostats:disordered_tracing.py:435 [test_image] : Disordered tracing of grain 0 failed. Consider raising an issue on GitHub. Error:
Traceback (most recent call last):
  File "/home/neil/work/git/hub/AFM-SPM/TopoStats/topostats/tracing/disordered_tracing.py", line 421, in trace_image_disordered
    print(f"\n{grain_crop.stats=}\n")
               ^^^^^^^^^^^^^^^^
  File "/home/neil/work/git/hub/AFM-SPM/TopoStats/topostats/grains.py", line 349, in stats
    return self._stats
           ^^^^^^^^^^^
AttributeError: 'GrainCrop' object has no attribute '_stats'
ERROR    topostats:disordered_tracing.py:435 [test_image] : Disordered tracing of grain 1 failed. Consider raising an issue on GitHub. Error:
Traceback (most recent call last):
  File "/home/neil/work/git/hub/AFM-SPM/TopoStats/topostats/tracing/disordered_tracing.py", line 421, in trace_image_disordered
    print(f"\n{grain_crop.stats=}\n")
               ^^^^^^^^^^^^^^^^
  File "/home/neil/work/git/hub/AFM-SPM/TopoStats/topostats/grains.py", line 349, in stats
    return self._stats
           ^^^^^^^^^^^
AttributeError: 'GrainCrop' object has no attribute '_stats'
FAILED                                                                                                         [ 50%]

====================================================== FAILURES ======================================================
```

I think this is down to the fact that the test loads `graincrops` from a pickle
(`tests/resources/example_catenanes_graincrops.pkl` or `tests/resources/example_rep_int_graincrops.pkl`) which need
updating as a consequence.

Noting the comment within this function about not blindly updating these I'm checking with you.

The expected items (`expected_disordered_crop_data` and `expected_all_images`) probably won't need updating need
updating I don't think, rather it is the two `graincrops` pickles mentioned above that will need updating to account for
the newly added attributes, so far `stats` and `height_profiles` but this is likely to expand as we work through moving
everything that was previously in dictionaries into the `GrainCrop` class.

Does that make sense or have I mimsunderstood how these are setup?

SylviaWhittle, 2025-04-02:
This makes sense to me, and you're all good for updating the pickle. the comment about not blindly updating them is just to caution against seeing that there is a discrepancy and then updating it without looking into what changed, but with this situation you know exactly what is changing so it's fine 😄 

I think keeping disordered tracing stats inside `stats` in `GrainCrop` objects seems appropriate 👍 

I don't see any reason not to proceed with this 👍 thanks @ns-rse!

(as always let me know if I've not addressed something or you think I might be misunderstanding the situation)

ns-rse, 2025-04-02:
Thanks for the quick response @SylviaWhittle I'll go ahead and update these.

Although I've been pondering on this and wonder if rather than loading pickles we might be able to construct these as
fixtures in `conftest.py` loading the underlying arrays from `.npy` files and instantiating the `GrainCrop` objects
anew. This would mean that longer term as I work through adding more attributes to the `GrainCrop` class we wouldn't
have to continually update the `.pkl` (things like `GrainCrop.stats` and `GrainCrop.heigh_profiles` could be calculated
on loading). It was this sort of thing that I had reservations about when these pickles were first introduced.

SylviaWhittle, 2025-04-02:
Yeah... Pickles are irking me. The lack of transparency is frustrating. I'd say go for it, see how hard it is to turn them into a fixture, though I'd be a little apprehensive if it causes a significant development slowdown when adding new things versus a pickle.

The conflict between ease of development and comprehensiveness in code is eternal. I keep sliding about on that scale.

ns-rse, 2025-04-03:
I checked back anyway :)

Cool, thanks, I'll see if I can switch the current pickles to fixtures first this morning and can then proceed with
adding new attributes. Currently the phrase "_One step-forward, two-steps back_" comes to mind! Its all progress though.

closed 2025-12-18 by ns-rse