AFM-SPM/TopoStats #1126: [Bug]: Grainstats entry point failure due to unfound "grain_masks" in .topostats

MaxGamill-Sheffield, 2025-04-03:
### Checklist

- [x] Find the offending file in the output. If processing halts, re-run analysis with `topostats --core 1 process`.
- [x] Describe the bug.
- [x] Include the configuration file.
- [x] Copy of the log-file from running with `topostats --log-level debug <command>`.
- [x] The exact command that failed. This is what you typed at the command line, including any options.
- [x] TopoStats version, this is reported by `topostats --version`
- [x] Operating System and Python Version

### Describe the bug

Grainstats (and maybe a few other) entry points fail due to missing "grain_masks" in .topostats files which should reference the new "grain_tensors".

### Copy of the log-file from running with `topostats --log-level debug <command>`

(main) Max:TopoStats Maxgamill$ topostats -c data/multiclass.yaml grainstats
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] The YAML configuration file is valid.
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] The YAML plotting configuration file is valid.
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] Configuration run options are consistent, processing can proceed.
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] Configuration file loaded from      : data/multiclass.yaml
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] Scanning for images in              : data
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] Output directory                    : data
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] Looking for images with extension   : .topostats
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] Images with extension .topostats in data : 1
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] Thresholding method (Filtering)     : std_dev
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] Thresholding method (Grains)        : absolute
[Thu, 03 Apr 2025 10:26:46] [INFO    ] [topostats] Extracting image from data/processed/20240610_6ng_puc19_50ng_HSf1_mgni.0_00009.topostats
10:26:46 | INFO |topostats.py:topostats:load_topostats:39 | Loading image from : data/processed/20240610_6ng_puc19_50ng_HSf1_mgni.0_00009.topostats
10:26:46 | INFO |topostats.py:topostats:load_topostats:48 | [20240610_6ng_puc19_50ng_HSf1_mgni.0_00009] TopoStats file version : 0.2
Traceback (most recent call last):
  File "/opt/miniconda3/envs/main/bin/topostats", line 8, in <module>
    sys.exit(entry_point())
  File "/opt/miniconda3/envs/main/lib/python3.10/site-packages/topostats/entry_point.py", line 1214, in entry_point
    args.func(args)
  File "/opt/miniconda3/envs/main/lib/python3.10/site-packages/topostats/run_modules.py", line 511, in grainstats
    all_scan_data.get_data()
  File "/opt/miniconda3/envs/main/lib/python3.10/site-packages/topostats/io.py", line 786, in get_data
    self.grain_masks = data["grain_masks"]
KeyError: 'grain_masks'

### Include the configuration file

# Config file generated 2025-03-26 11:01:06
# # For more information on configuration and how to use it:
# https://afm-spm.github.io/TopoStats/main/configuration.html
base_dir: ./data # Directory in which to search for data files
output_dir: ./data # Directory to output results to
log_level: info # Verbosity of output. Options: warning, error, info, debug
cores: 2 # Number of CPU cores to utilise for processing multiple files simultaneously.
file_ext: .topostats # File extension of the data files.
loading:
  channel: Height # Channel to pull data from in the data files.
  extract: raw # Array to extract when loading .topostats files.
filter:
  run: true # Options : true, false
  row_alignment_quantile: 0.5 # lower values may improve flattening of larger features
  threshold_method: std_dev # Options : otsu, std_dev, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    below: 10.0 # Threshold for data below the image background
    above: 1.0 # Threshold for data above the image background
  threshold_absolute:
    below: -1.0 # Threshold for data below the image background
    above: 1.0 # Threshold for data above the image background
  gaussian_size: 1.0121397464510862 # Gaussian blur intensity in px
  gaussian_mode: nearest # Mode for Gaussian blurring. Options : nearest, reflect, constant, mirror, wrap
  # Scar remvoal parameters. Be careful with editing these as making the algorithm too sensitive may
  # result in ruining legitimate data.
  remove_scars:
[... 247 lines cut ...]
superqt==0.7.3
tensorboard==2.19.0
tensorboard-data-server==0.7.2
tensorflow==2.19.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==3.0.0
threadpoolctl==3.6.0
tifffile==2025.3.30
tokenize_rt==6.1.0
tomli==2.2.1
tomlkit==0.13.2
toolz==1.0.0
topoly==1.0.4
topostats @ file:///Users/Maxgamill/Desktop/Uni/PhD/topo_main/TopoStats
tornado==6.4.2
tqdm==4.67.1
traitlets==5.14.3
typing_extensions==4.13.0
tzdata==2025.2
urllib3==2.3.0
virtualenv==20.30.0
wcwidth==0.2.13
Werkzeug==3.1.3
wrapt==1.17.2
wurlitzer==3.1.1

MaxGamill-Sheffield, 2025-04-03:
This seems like a bigger problem than I first thought, all changed code is on `maxgamill-sheffield/1126-entrypoint-failure`:

1. The simple fix of swapping "grain_masks" to "grain_tensors" in `io.py::get_data:785`:
```
if self.extract in ("grainstats"):
    self.grain_masks = data["grain_tensors"]
```
2. Then the `grain_crops` output is missing from `run_modules::grainstats:526`:
```
for img, result, height_profiles, grain_crops in pool.imap_unordered(
```
3. Depending on the intention of cleaning the image dictionary in `io.py::get_data:805`, default values might need to be added, for example, when you have processed upto the `Grains` step, then remember you want to process the `Grainstats` steps, the values that are popped (e.g. "disordered_traces") may not be present in the dictionary. This would need to be applied throughout the `io.py::clean_dict:805`:
```
if self.extract in ["grainstats"]:
    img_dict.pop("disordered_traces", None)
    img_dict.pop("grain_curvature_stats", None)
    img_dict.pop("height_profiles", None)
    img_dict.pop("nodestats", None)
    img_dict.pop("ordered_traces", None)
    img_dict.pop("splining", None)
```
4. Errors then come from passing the "grain_tensors" into the `processing.py::process_grainstats:1486` in the `if` statement, and in the `image_grain_crops=` statement shortly after. This is I think what should be the `grain_crops` class object as `run_grainstats` converts this to a dictionary using `image_grain_crops.__dict__.items()` (`processing.py::run_grainstats:369`) which fails as this is already a dictionary in the `.topostats` file.

@ns-rse and @SylviaWhittle, do y'all have any suggestions on how to load the grains from the `.topostats` file and keep it in the `GrainCrops` format for the proceeding steps (grainstats etc)?

ns-rse, 2025-04-04:
Hi @MaxGamill-Sheffield

This is I think part of what is being addressed by #1102 which I've broken donw into #1112 to #1117.

Each of these I intend to add attributes for the various objects to `GrainCrop` which are currently returned as dictionaries, leaving those in place so that the whole pipeline works and people can use it.

Once we have all of the attributes as properties of `GrainCrop` we can then write these out to `.topostats` and bump the version to `0.3`.

@SylviaWhittle and I had given some thought about how to read `.topostats` and whether the reconstruction of `ImageGrainCrops` with `above`/`below` attributes, each of which is `GrainCropsDirection` which holds the `full_mask_tensor` and a dictionary of `GrainCrops` objects.

There were two options we considered...

1. AFMReader reconstructs these classes from the HDF5 that is loaded.
2. AFMReader loads the HDF5 and passes it back to TopoStats to reconstruct the classes.

We felt that the latter option was preferabler to avoid introducing circular dependencies whereby TopoStats depends on AFMReader but AFMReader also depends on TopoStats.

Thus we need to work through #1112 to #1117 to make sure that what TopoStats internal representation of these classes and how it writes `.topostats` files contains everything it needs and that these can then be loaded back in to the same as when they are read (until recently loading in `.topostats` gave a different structure to the internal structure, see [AFMReader #112](https://github.com/AFM-SPM/AFMReader/pull/112)).

_Then_ we can correct this issue with `topostats grainstats` and add entry points to all the other steps.