AFM-SPM/TopoStats #1161: [Bug]: UnboundLocalError of local variable 'val' from get_layer_value after running "topostats process" (topoStats version 2.3.1)

Malisa-Vittoria, 2025-05-22:
### Checklist

- [x] Find the offending file in the output. If processing halts, re-run analysis with `topostats --core 1 process`.
- [x] Describe the bug.
- [x] Include the configuration file.
- [x] Copy of the log-file from running with `topostats --log-level debug <command>`.
- [x] The exact command that failed. This is what you typed at the command line, including any options.
- [x] TopoStats version, this is reported by `topostats --version`
- [x] Operating System and Python Version

### Describe the bug

Error appears when processing AFM images by using the command "topostats process" and the error appears when loading AFM images by using the command "topostats load". Error message in both cases suggest that there is an issue with the assignment of the local variable 'val' that is referenced before assignment.

### Copy of the log-file from running with `topostats --log-level debug <command>`

[20240422-2025-05-22-11-04-55.log](https://github.com/user-attachments/files/20383932/20240422-2025-05-22-11-04-55.log)

### Include the configuration file

[config_error_message.txt](https://github.com/user-attachments/files/20383951/config_error_message.txt)

### TopoStats Version

2.1.2

### Python Version

3.1

### Operating System

Windows

### Python Packages

absl-py==2.2.2
AFMReader==0.0.4
art==6.5
asttokens==3.0.0
astunparse==1.6.3
biopython==1.85
certifi==2025.4.26
charset-normalizer==3.4.2
cheap_repr==0.5.2
colorama==0.4.6
contourpy==1.3.2
cycler==0.12.1
decorator==5.2.1
docstring_parser==0.16
et_xmlfile==2.0.0
exceptiongroup==1.3.0
executing==2.2.0
flatbuffers==25.2.10
fonttools==4.58.0
gast==0.6.0
google-pasta==0.2.0
grpcio==1.71.0
h5py==3.13.0
idna==3.10
igor2==0.5.12
imageio==2.37.0
ipython==8.36.0
jedi==0.19.2
joblib==1.5.0
keras==3.10.0
kiwisolver==1.4.8
lazy_loader==0.4
libclang==18.1.1
llvmlite==0.44.0
loguru==0.7.3
magicgui==0.10.0
Markdown==3.8
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.9.4
matplotlib-inline==0.1.7
mdurl==0.1.2
ml_dtypes==0.5.1
namex==0.0.9
networkx==3.4.2
numba==0.61.2
numpy==1.26.4
numpyencoder==0.3.0
openpyxl==3.1.5
opt_einsum==3.4.0
optree==0.15.0
packaging==25.0
pandas==2.2.3
parso==0.8.4
pillow==11.2.1
prompt_toolkit==3.0.51
protobuf==5.29.4
psutil==5.9.8
psygnal==0.13.0
pure_eval==0.2.3
pyconify==0.2.1
Pygments==2.19.1
pyparsing==3.2.3
pyspm==0.6.2
python-dateutil==2.9.0.post0
pytz==2025.2
PyYAML==6.0.2
QtPy==2.4.3
requests==2.32.3
rich==14.0.0
ruamel.yaml==0.18.10
ruamel.yaml.clib==0.2.12
schema==0.7.7
scikit-image==0.25.2
scikit-learn==1.6.1
scipy==1.15.3
seaborn==0.13.2
six==1.17.0
skan==0.12.2
snoop==0.6.0
stack-data==0.6.3
superqt==0.7.3
tensorboard==2.19.0
tensorboard-data-server==0.7.2
tensorflow==2.19.0
tensorflow-io-gcs-filesystem==0.31.0
termcolor==3.1.0
threadpoolctl==3.6.0
tifffile==2025.5.10
toolz==1.0.0
topoly==1.1.0
topostats==2.3.1
tqdm==4.67.1
traitlets==5.14.3
typing_extensions==4.13.2
tzdata==2025.2
urllib3==2.4.0
wcwidth==0.2.13
Werkzeug==3.1.3
win32_setctime==1.2.0
wrapt==1.17.2

ns-rse, 2025-05-23:
Hi @Malisa-Vittoria 

Thanks for taking the time to report this issue. We'll see if we can investigate over the coming week. If you are able to share the image file with us that would greatly assist in our investigations. Feel free to mail a copy to [topostats@sheffield.ac.uk](mailto:topostats@sheffield.ac.uk) or use another file sharing service.

You've indicated that you found this problem with `TopoStats==2.1.2`. Would you be able to see if the error occurs with the most recent version of `TopoStats==2.3.0` please?

You can upgrade the installed version within your virtual environment by activating the environment and then running...

```
pip install --upgrade topostats
```

ns-rse, 2025-05-23:
Reading through the output in #1162 suggests this is actually a problem with [AFMReader](https://github.com/AFM-SPM/AFMReader) within [`AFMReader/spm.py`](https://github.com/AFM-SPM/AFMReader/blob/main/AFMReader/spm.py).

The line numbers in the error log don't appear to align with the current status of the `spm.py`.

It would therefore be worth upgrading AFMReader too...

```
pip install --upgrade AFMReader
```

ns-rse, 2025-05-27:
Hi @Malisa-Vittoria that was my mistake sorry. I forgot I had an issue blocking my release of `v0.0.5` which I resolved over the weekend. I've just made a [new release of v0.0.5](https://pypi.org/project/AFMReader/#history).

You should have been able to install the development version though. Did you have any problems with that?

Malisa-Vittoria, 2025-05-27:
> Hi [@Malisa-Vittoria](https://github.com/Malisa-Vittoria) that was my mistake sorry. I forgot I had an issue blocking my release of `v0.0.5` which I resolved over the weekend. I've just made a [new [...]

Hi! 
Thank you for looking into this issue! 
I could upgrade the AFMReader to 0.0.5 with Topostats version 2.3.1 and Topostats version 2.3.0. However, I still get the same error message. I also tried to run "topostats process" and "topostats load" and it appears in both commands. 
Unfortunately, it was not possible to select the right version of TopoStats in the reporting section of Github. This is the reason why I wrote the TopoStats version in the header of the issue reporting. 
I am not sure, whether this error arises from the images that I am looking at, i.e. amyloid fibrils which I have sent to you by Email or whether it is script related.

ns-rse, 2025-05-28:
Ah, ok we need to update out issue template.

I've started looking into this and the problem stems from the [pySPM](https://github.com/scholi/pySPM) package which [AFMReader](https://github.com/AFM-SPM/TopoStats) uses to load `.spm` files.

```
# Load the pyspm.Bruker library
from pyspm import Bruker

# Load one of the files and list the available channels
scan = Bruker("20240422_SampleB_ROI1_1V_2x2um.spm")
scan.list_channels()

# List the layers
scan.layers
```

This gives the following channels and layers...

```
Channels
========
S [ZSensor] "Height Sensor"
S [PeakForceError] "Peak Force Error"
S [Stiffness] "DMTModulus"
S [LogStiffness] "LogDMTModulus"
S [Adhesion] "Adhesion"
S [Deformation] "Deformation"
S [LockIn2AmpInput] "Amplitude2" (MFM)
S [] "Potential" (MFM)
```

Layers which is a list, I've annotated it with `# Layer <number>`

```

scan.layers=[{b'Data offset': [b'80960'], # Layer 0
              b'Data length': [b'1048576'],
              b'Bytes/pixel': [b'4'],
              b'Start context': [b'OL'],
              b'Data type': [b'AFM'],
              b'Plane fit': [b'0 0 0 5'],
              b'Frame direction': [b'Up'],
              b'Capture start line': [b'0'],
              b'Color Table Index': [b'12'],
              b'Relative frame time': [b'7361.78'],
              b'Invalid Data Flag': [b'None'],
              b'Invalid Data Fill': [b'None'],
              b'Samps/line': [b'512'],
              b'Number of lines': [b'512'],
              b'Aspect Ratio': [b'1:1'],
              b'Scan Size': [b'2 2 ~m'],
              b'Scan Line': [b'Main'],
              b'Line Direction': [b'Retrace'],
              b'Highpass': [b'0'],
              b'Lowpass': [b'0'],
              b'Realtime Planefit': [b'Line'],
              b'Offline Planefit': [b'None'],
              b'Valid data start X': [b'0'],
              b'Valid data start Y': [b'0'],
              b'Valid data len X': [b'512'],
              b'Valid data len Y': [b'512'],
              b'Tip x width correction factor': [b'1'],
              b'Tip y width correction factor': [b'1'],
              b'Tip x width correction factor sigma': [b'1'],
              b'Tip y width correction factor sigma': [b'1'],
              b'@2:Image Data': [b'S [ZSensor] "Height Sensor"'],
              b'@Z magnify': [b'C [2:Z scale] 0.001382432'],
              b'@2:Z scale': [b'V [Sens. ZsensSens] (0.00000000572205 V/LSB) 24.57600 V'],
              b'@2:Z offset': [b'V [Sens. ZsensSens] (0.00000000572205 V/LSB) 0.7432500 V']},
             {b'Data offset': [b'1129536'],  # Layer 1
[... 395 lines cut ...]
flipped = np.flipud(np.asarray(data.pixels))
# Now plot using TopoStats
plotting = plottingfuncs.Images(data=flipped, output_dir="./", filename="20240422_SampleB_ROI1_1V_2x2um.spm")
plotting.save_figure("topostats_plot.png")
```

Which gives us the following image...

[image]

## What to do?

This is perhaps a new format/type of `.spm` that the authors of [pySPM](https://github.com/scholi/pySPM) haven't encountered. Because of a few other issues that I've encountered with the package recently I've been making contributions to that package and am now able to contribute directly.

I'll aim to get this addressed in pySPM and fixed there rather than patching in TopoStats. My time working on TopoStats is up for this week so I'll have to do that next week I'm afraid.

In the mean time if you wanted to add the suggested line to your installed version of pySPM that should get you going. You can alternatively clone their repository, make the change in the source code and `pip install -e .` . This is the same as the instructions in the [TopoStats Documentation](https://afm-spm.github.io/TopoStats/main/installation.html#cloning-the-repository-and-installing) to install from source but you would use...

```
cd ~/where/to/clone
git clone git@github.com:scholi/pySPM.git
cd pySPM
# Edit the file pySPM/Bruker.py to include the lines mentioned above and save the file then install with...
pip install -e .
```

ns-rse, 2025-05-28:
Oh yes, I forgot, you'd have to generate a configuration file (see `topostats create-config --help`) and edit the `channel: "Height Sensor"` to then use TopoStats to process images.

Malisa-Vittoria, 2025-05-28:
Thank you very much for the help! The detailed explanation helped me to understand the issue. 

Yes, the channel in my configuration file was set wrong (because it didn't exist in the raw data). Thank you for correcting it! 

I had only issues in loading and processing the spm files because I could process images in gwy format. So, as suggested, I will modify the source code in pySPM to be able to process spm files with TopoStats. The AFM image you showed is the one I have sent to you. 

Thank you again!